syntalos/syntalos #2: can not add modules

Wulin-Tan, 2023-12-27:
Hi
syntalos is a great tool. 
after I open the software and click "Add Modules", there is no modules for me to select.
Any suggestions?
Thank you.

ximion, 2023-12-27:
Hi!
How did you install Syntalos? (packages, self-compiled, Flatpak, ...)
Can you paste the output of the log displayed when you open `Diagnostics → Module Loader` here, as well as copy the system information from `Diagnostics → System` and paste it here as well? (the latter window has a "Copy to Clipboard" button in the bottom left :-)

ximion, 2023-12-27:
Hi!
Thank you for using Syntalos! :-)
How did you install it? (packages, self-compiled, Flatpak, ...)
Can you paste the output of the log displayed when you open `Diagnostics → Module Loader` here (in the main menu of the Syntalos window), as well as copy the system information from `Diagnostics → System` and paste it here as well? (the latter window has a "Copy to Clipboard" button in the bottom left :smile:)

Wulin-Tan, 2023-12-28:
Hi, @ximion 
I installed the app by Flatpak.
when I run Diagnostics → Module Loader, it shows:
Module Loader Log
Loading modules from:  
Loaded 0 modules.

when I run Diagnostics → System, it shows:
Syntalos 1.0.0 | KDE Flatpak runtime (Kernel: linux 5.15.0-91-generic)
Software: Qt 5.15.10 | OpenCV 4.8.0 | FFMpeg 6.0 | Python 3.10
Sandbox: None detected
Architecture: x86_64 | CPU: 12th Gen Intel(R) Core(TM) i9-12900H | OpenGL: 4.6 (Compatibility Profile) Mesa 23.3.0 (git-1fbdd37d4c)
Misc: Missing hardware support package.

hope to get further suggestion, since this tool is great and important to my project.
Thank you. @ximion

ximion, 2023-12-28:
Very interesting! There is something super odd about your configuration, as you say you installed the app from the store using Flatpak, and yet the Syntalos you launched was installed from a Debian package.
Can you run: `sudo apt purge syntalos*` in a terminal and see if that resolves it?
Alternatively, you can also run `flatpak run io.github.bothlab.syntalos` in a terminal to make absolutely sure you are launching the store app, and not accidentally a differently installed version.

Then check again, modules should be loaded. The "Module Loader" diagnostics window should also give you a path where modules are loaded from, it should look like this:
``` 
Loading modules from: /app/lib/syntalos/modules
Loaded 21 modules.
```

Wulin-Tan, 2023-12-29:
Hi @ximion 

I tried the way you suggested, but still it could not load modules:

sudo apt purge syntalos* , it gives:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package syntalos*
E: Couldn't find any package by glob 'syntalos*'

flatpak run io.github.bothlab.syntalos, it can start the app, but gives followings:

Gtk-Message: 10:37:18.777: Failed to load module "canberra-gtk-module"
Gtk-Message: 10:37:18.777: Failed to load module "canberra-gtk-module"
qt.qpa.qgnomeplatform: Could not find color scheme  ""
Qt: Session management error: Could not open network socket
Switched icon theme to breeze-dark
engine: Application data directory: /home/tanwulin/.var/app/io.github.bothlab.syntalos/data/DraguhnLab/Syntalos
modulelibrary: Loading modules from location:

ximion, 2023-12-29:
Okay, you may have found a bug, but I can not reproduce it at all. You are reaching a state that should not be possible to reach in the code...
Can you help me to investigate?

Please give me the output of `flatpak --version` and `flatpak run --command=sh io.github.bothlab.syntalos -c export` (in the latter case the output may be a bit long, if you want to you can attach it as text file).

Also, just to test a wild guess of mine, what happens if you run this command, are any modules loaded or does it show you a path where they are loaded from at least? => `flatpak run --env=LANG=C.UTF-8 io.github.bothlab.syntalos`

Wulin-Tan, 2023-12-29:
Hi, @ximion 

flatpak --version =>
Flatpak 1.6.5

flatpak run --command=sh io.github.bothlab.syntalos -c export =>
export AT_SPI_BUS_ADDRESS="unix:path=/run/user/1000/at-spi-bus"
export COLORTERM="truecolor"
export CONDA_DEFAULT_ENV="base"
export CONDA_EXE="/home/tanwulin/anaconda3/bin/conda"
export CONDA_PREFIX="/home/tanwulin/anaconda3"
export CONDA_PROMPT_MODIFIER="(base) "
export CONDA_PYTHON_EXE="/home/tanwulin/anaconda3/bin/python"
export CONDA_SHLVL="1"
export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"
export DBUS_SYSTEM_BUS_ADDRESS="unix:path=/run/dbus/system_bus_socket"
export DESKTOP_SESSION="ubuntu"
export DISPLAY=":99.0"
export FLATPAK_ID="io.github.bothlab.syntalos"
export FLATPAK_SANDBOX_DIR="/home/tanwulin/.var/app/io.github.bothlab.syntalos/sandbox"
export GDMSESSION="ubuntu"
export GI_TYPELIB_PATH="/app/lib/girepository-1.0"
export GJS_DEBUG_OUTPUT="stderr"
export GJS_DEBUG_TOPICS="JS ERROR;JS LOG"
export GNOME_DESKTOP_SESSION_ID="this-is-deprecated"
export GNOME_SHELL_SESSION_MODE="ubuntu"
export GNOME_TERMINAL_SCREEN="/org/gnome/Terminal/screen/7a8a6110_472d_4178_ba96_18e1b9c25f0a"
export GNOME_TERMINAL_SERVICE=":1.90"
export GPG_AGENT_INFO="/run/user/1000/gnupg/S.gpg-agent:0:1"
export GST_PLUGIN_PATH="/app/lib/gstreamer-1.0:/usr/lib/extensions/gstreamer-1.0:/usr/lib/x86_64-linux-gnu/gstreamer-1.0"
export GST_PLUGIN_SYSTEM_PATH="/app/lib/gstreamer-1.0:/usr/lib/extensions/gstreamer-1.0:/usr/lib/x86_64-linux-gnu/gstreamer-1.0"
export GTK_MODULES="gail:atk-bridge"
export HOME="/home/tanwulin"
export IM_CONFIG_PHASE="1"
export INVOCATION_ID="8f3b5458d6f74ce2867d2d7d919ef3ab"
export JOURNAL_STREAM="8:47586"
export KDE_FORK_SLAVES="1"
export LANG="en_US.UTF-8"
export LC_ADDRESS="en_US.UTF-8"
export LC_IDENTIFICATION="en_US.UTF-8"
export LC_MEASUREMENT="en_US.UTF-8"
export LC_MONETARY="en_US.UTF-8"
export LC_NAME="en_US.UTF-8"
export LC_NUMERIC="en_US.UTF-8"
export LC_PAPER="en_US.UTF-8"
export LC_TELEPHONE="en_US.UTF-8"
export LC_TIME="en_US.UTF-8"
export LESSCLOSE="/usr/bin/lesspipe %s %s"
export LESSOPEN="| /usr/bin/lesspipe %s"
export LOGNAME="tanwulin"
export LS_COLORS="rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:"
export MANAGERPID="1131"
export OLDPWD
export PAPERSIZE="letter"
export PATH="/app/bin:/usr/bin:/usr/sbin"
export PWD="/home/tanwulin"
export QML2_IMPORT_PATH="/app/lib64/qml:/app/lib/qml"
export QT_ACCESSIBILITY="1"
export QT_IM_MODULE="ibus"
export QT_PLUGIN_PATH="/app/lib64/plugins:/app/lib/plugins:/usr/share/runtime/lib/plugins"
export SESSION_MANAGER="local/tanwulin-Default-string:@/tmp/.ICE-unix/1385,unix/tanwulin-Default-string:/tmp/.ICE-unix/1385"
export SHELL="/bin/sh"
export SHLVL="2"
export SSH_AGENT_PID="1347"
export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"
export TERM="xterm-256color"
export USER="tanwulin"
export USERNAME="tanwulin"
export VTE_VERSION="6003"
export WINDOWPATH="2"
export XAUTHORITY="/run/user/1000/Xauthority"
export XDG_CACHE_HOME="/home/tanwulin/.var/app/io.github.bothlab.syntalos/cache"
export XDG_CONFIG_DIRS="/app/etc/xdg:/etc/xdg"
export XDG_CONFIG_HOME="/home/tanwulin/.var/app/io.github.bothlab.syntalos/config"
export XDG_CURRENT_DESKTOP="ubuntu:GNOME"
export XDG_DATA_DIRS="/app/share:/usr/share:/usr/share/runtime/share:/run/host/user-share:/run/host/share"
export XDG_DATA_HOME="/home/tanwulin/.var/app/io.github.bothlab.syntalos/data"
export XDG_MENU_PREFIX="gnome-"
export XDG_RUNTIME_DIR="/run/user/1000"
export XDG_SESSION_CLASS="user"
export XDG_SESSION_DESKTOP="ubuntu"
export XDG_SESSION_TYPE="x11"
export XMODIFIERS="@im=ibus"
export _="/usr/bin/flatpak"
export _CE_CONDA=""
export _CE_M=""
export __EGL_EXTERNAL_PLATFORM_CONFIG_DIRS="/etc/egl/egl_external_platform.d:/usr/lib/x86_64-linux-gnu/GL/egl/egl_external_platform.d:/usr/share/egl/egl_external_platform.d"

flatpak run --env=LANG=C.UTF-8 io.github.bothlab.syntalos => (it just launched the app and the terminal showed the info as follows, but didn't  show any path)

Gtk-Message: 11:47:47.287: Failed to load module "canberra-gtk-module"
Gtk-Message: 11:47:47.287: Failed to load module "canberra-gtk-module"
qt.qpa.qgnomeplatform: Could not find color scheme  ""
Qt: Session management error: Could not open network socket
Switched icon theme to breeze-dark
engine: Application data directory: /home/tanwulin/.var/app/io.github.bothlab.syntalos/data/DraguhnLab/Syntalos
modulelibrary: Loading modules from location:

ximion, 2023-12-29:
Thank you! I am pretty sure you are running Ubuntu 20.04 and that the issue is caused by a small incompatibility with older Flatpak versions that we didn't notice because all systems we tested on were on Ubuntu 22.04.
Can you upgrade your system to Ubuntu 22.04 LTS? I also think we can address this in Syntalos, so I'll try to roll out an update on Flathub soon, so we don't see this odd behavior on older Ubuntu/Flathub versions :-)
Thanks for reporting this issue!

closed 2023-12-29 by ximion

ximion, 2023-12-29:
I applied the change to the copy on Flathub via https://github.com/flathub/io.github.bothlab.syntalos/pull/5, so with a bit of luck you should get an update soon that resolves this for old Flatpak versions.
Regardless, a system upgrade is probably a good idea, if it's possible / permitted.

Thank you for using Syntalos!

Wulin-Tan, 2023-12-30:
Hi, @ximion 
when I upgraded to ubuntu 22.04, modules can be added and syntalos works perfectly!
Thank you.

commit 0dfcb493 (2023-12-29): Fix module loader on older versions of Flatpak / Ubuntu 20.04

Resolves: #2