syntalos/syntalos #4: RHA intan is possible?

anacastropsico, 2024-03-19:
It would be possible to use the software with intan amplifier demo 1.03, inta's oldest software, used for the RHA board, not compatible with the RHX software.

ximion, 2024-03-19:
I am afraid not, as internally Syntalos uses Intan's RHX code - so we support everything Intan officially does, but not more.
Of course, a module for RHA could be added, but given that the hardware is discontinued and it's comparatively high-effort to add such a module, I don't think that will happen. Sorry.

There is some options: You could switch to an RHD chip, which of course comes with some costs, but is absolutely worth it - signal-to-noise is a lot better with the digital version, and we phased out the last RHA station in our lab last year.
Alternatively, Syntalos can emit a control pulse to an external application to start/stop it when the Syntalos experiment is started/stopped. Using that would require significantly less coding work than making the old RHA code a new module, but Synchronization would not be as perfect as when it was integrated, and you would not benefit from the shared data storage and other Syntalos benefits.

It's a pain that the "RHD USB interface board" is no longer available, as now switching from RHA to RHD is a lot more expensive than before :-/

anacastropsico, 2024-03-19:
Thank you for your willingness to respond and give suggestions, I will try to implement

ximion, 2024-03-19:
I would have written such a module, but given that you can't buy RHA hardware anymore and that I currently also have none to test it with, that would not have been a good idea.

You can reach out to Intan as well, they were generally very helpful with requests - potentially they could add support to the RHX software, which would flow back into Syntalos as well :-)

anacastropsico, 2024-03-19:
Good idea, I'll contact you, thans!

ximion, 2024-03-19:
Good luck with your experiments! Don't hesitate to file another bug if you find an issue or write in the discussions tab! https://github.com/bothlab/syntalos/discussions

closed 2024-03-19 by ximion