syntalos/syntalos #5: DAQ not outputting signal when using Syntalos

tamai8810, 2024-04-26:
Hello, Mattias

I am currently using Syntalos for experiments with Miniscope.
While the signal output from the DAQ terminal during frame acquisition [(this)](http://miniscope.org/index.php/Data_Acquisition_Box) works correctly with PoMiDAQ, it does not output at all with Syntalos.

↓Sync Output port toggles its output on each frame acquisition.
[image]

↓Sync Output when PoMiDAQ handles miniscope
[image]

↓Sync Output when Syntalos handles miniscope
[image]

Do you know what might be causing this issue?

Best regards,
Hayato

closed 2024-04-26 by ximion

ximion, 2024-04-26:
So, here's the thing: Syntalos in its master branch is currently undergoing a lot of changes to make its API more versatile (and suitable for more applications), improve scripting latency dramatically and more. It *can* definitely be used for data acquisition right now, but we aren't even done with our own testing and validation of the new version yet. So, you'd be a beta tester :sweat_smile: 

I fixed the issue in PoMiDAQ though, which means you wouldn't have to use the latest Syntalos version. But you will have to rebuild PoMiDAQ from source, install it, and make sure that Syntalos' `miniscope` module uses the latest version (best done by simply removing any older version from the linker search path).

Can you do that? Otherwise I'll have to look into maybe releasing a new version of PoMiDAQ sooner (a new Syntalos version will still take a while due to the volume of changes and testing needed).

tamai8810, 2024-04-28:
Thank you for your prompt response.

Thanks to your support, I believe I’m close to resolving the issue. However, as mentioned in the PoMiDAQ discussion [(here),](https://github.com/bothlab/pomidaq/issues/13) my project requires recording video with frame division using Syntalos. 

I’m concerned that the current version of PoMiDAQ may not support this function. Could you please share your thoughts on whether this is possible or suggest any alternatives?

I’m going to rebuild PoMiDAQ after the decision. 

I appreciate your guidance on this matter.

Best regards,
Hayato

ximion, 2024-04-28:
Ah, sorry, you probably misunderstood me :-)
You can continue to use Syntalos without any changes. PoMiDAQ provides a shared library, `libminiscope`, which can be used to embed Miniscope functionality into any application, Syntalos being one of them. The change for your issue is in `libminiscope`.

How did you install Syntalos? How did you install PoMiDAQ? If you instaleld Syntalos on a Debian or Ubuntu system from the deb packages, all you need to do is to install these PoMiDAQ packages which contain the fix:
[Ubuntu Package.zip](https://github.com/bothlab/syntalos/files/15142124/Ubuntu.Package.zip)
[Debian Package.zip](https://github.com/bothlab/syntalos/files/15142137/Debian.Package.zip)

The Syntalos should work as you expect it to work :-)
If you installed Syntalos from Flathub, things are a bit more complicated - in that case, let me know, I think I can actually submit a change to Flathub that will resolve this for you too, so you won't have to compile any code on your own :-)

tamai8810, 2024-04-30:
Thank you very much for your assistance; it has been invaluable.

To my dismay, I have installed Syntaros from Flathub. 
Could you please advise me on the correct procedure to follow in this situation? 
Any specific steps or recommendations you could provide would be greatly appreciated.

Thank you once again for your kind help.

Best regards,

ximion, 2024-04-30:
> To my dismay, I have installed Syntaros from Flathub.

No worries! In this one case, I can provide an update, as it's a very easy, targeted change.

You should receive a Syntalos update today. If it isn't offered to you automatically in a few hours by your Software Center, you can also run `flatpak update` in a terminal manually to trigger it.

That's all that should be needed :-)

tamai8810, 2024-05-15:
Thank you for your kind instructions.

I am pleased to inform you that I have finally installed the .deb file on Ubuntu OS, and it is functioning correctly.

Best regards,

ximion, 2024-05-15:
Nice! This should also be fixed in the Flatpak version already :-)

Good luck with your experiments!