syntalos/syntalos #6: Frame rate issue

athp18, 2024-06-25:
Hi,
We are trying to sync up the UCLA miniscope with an Orbbec Femto Bolt for simultaneous data acquisition. The Femto Bolt is currently supported by Linux V4L in addition to OpenCV. We are facing two problems: (1) simultaneous recordings don't work (only the miniscope records, and it will stop after some time with a "camera disconnected" issue (2) even without the miniscope, the Femto Bolt records at roughly 7-10 fps (it should record ideally at ~30). Wondering if anyone has any thoughts or could help with this :)

ximion, 2024-06-26:
Hi!
Very cool to hear that you are trying Syntalos! I am currently travelling so it is a bit harder for me to look into issues and test things, however there is still a few things you can try!

How did you install Syntalos / which version are you using? Would you feel comfortable with compiling Syntalos from source, or would you rather want to avoid this?

On the multiple issues:
(1): Does the Miniscope quit with "camera disconnected" or does the camera do that? In case it's the Miniscope, can you try a different USB port? The Miniscope device is especially sensitive to data transfer rates and doesn't do well on USB hubs that have a lot of other devices attached. The "Diagnostics" menu has a way to list which USB hubs devices are attached to. Also check for issues with all cables. Did you get any "Unable to grab frame" error messages as well?

(2) The framerate issues are super annoying and usually tied to the camera requiring some specific settings to work. Can you play with the "Capture Format" setting a bit to see if that yields any change? I guess you are using the "Generic Camera" module, right? Unfortunately, the Femto Bolt does not seem to support the GenICam standard, otherwise there would be a simple solution to this issue by using the "TIS Camera" or the new "Aravis Camera" modules.

You can also try the V4L Workbench to see which settings combination gives you the expected framerate, and then mention it so I can have a look if there are any knobs that the Syntalos module should expose.

athp18, 2024-06-27:
Hi, I'm going to close this issue since it turns out its PC-specific (we used it on another PC recently and it ended up working fine). However, I was wondering - can we use Syntalos to simultaneously record depth frames (which the Orbbec Femto Bolt supports) along with the Miniscope? Not sure if syntalos currently supports that.

closed 2024-06-27 by athp18

ximion, 2024-07-10:
> However, I was wondering - can we use Syntalos to simultaneously record depth frames (which the Orbbec Femto Bolt supports) along with the Miniscope?

I assume not (yet?) as there is no standard interface for these devices... You would have to write a module for Syntalos for the specific device using its SDK to make this work.

I could have a quick look at this to see how easy that would be to maybe give you some pointers if you want to pursue this path (module creation is currently being documented but it will take a bit of time).

If there is a standard for 3D cameras like GenICam for industrial cameras, it would be a bit easier to support many devices at once, but such as thing likely does not exist yet...

athp18, 2024-07-10:
> I assume not (yet?) as there is no standard interface for these devices... You would have to write a module for Syntalos for the specific device using its SDK to make this work.
>
> I could have a [...]

Yes, please, I'd appreciate that a lot! I don't have any problems writing a module but pointers on getting this module to work properly with Syntalos would definitely be appreciated :) Thank you

ximion, 2024-07-11:
Biggest immediate advice is to use Git master, as it contains a new API for modules :-)

There is actually three ways to implement modules:
* Using Python via the Python interface. Pro: Can be written entirely in Python and use the full range of stuff in PyPI. Stable API. Con: Is written in Python and running in a separate process, hence you get higher latencies and slower code.
* Using C++ as an out-of-process executable via the `libsyntalos-mlink` library. Pro: Easy and safe to work with, module crashes will never bring Syntalos itself down. Stable API usable for 3rd-parties and even in proprietary code (soon!). C++ code will be very performant. In future, other languages that aren't C/C++ could be added (like D, Rust and Java). Con: Higher processing latency due to IPC overhead. Currently not very good GUI integration (work-in-progress).
* Native C++ module. Pro: All advantages of an mlink C++ module, plus no latency penalty. Very fine control over all aspects of the module. Extremely good GUI integration. Con: More complex to implement, modules should ideally ship with the Syntalos sources to ensure API compatibility. A module crash can bring down Syntalos. In rare cases, interference with other modules (some device libraries do evil black magic and need some special isolation to play nice with others).

You will most likely want a native C++ module for a high-volume data source device like this. There is an example module that can just be duplicated and extended: https://github.com/syntalos/syntalos/tree/master/modules/example-cpp
There is also a mock datasource module that can be used for inspiration: https://github.com/syntalos/syntalos/tree/master/modules/devel.datasource

However, what this all really needs is more comprehensive documentation. I am currently hard at work on a paper revision for this software, of which improving documentation is a part, so I will get to it eventually (developer docs have a little lower priority than user docs).

athp18, 2024-07-12:
> Biggest immediate advice is to use Git master, as it contains a new API for modules :-)
>
> There is actually three ways to implement modules:
>
> * Using Python via the Python interface. Pro: Can be [...]

Thank you! These are great resources!