syntalos/syntalos #7: Libvips build issue

athp18, 2024-07-10:
Hi! I'm trying to build Syntalos from source, and one error I keep getting when trying to compile the project is this: 

```
[12/1002] Compiling C++ object src/dat...los-datactl.so.1.2.0.p/vipsutils.cpp.o
FAILED: src/datactl/libsyntalos-datactl.so.1.2.0.p/vipsutils.cpp.o 
c++ -Isrc/datactl/libsyntalos-datactl.so.1.2.0.p -Isrc/datactl -I../src/datactl -I. -I.. -Isrc -I../src -I../src/datactl/include -Isrc/utils -I../src/utils -I/usr/local/include -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/fribidi -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/x86_64-linux-gnu -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/lib/x86_64-linux-gnu/hdf5/serial/include -I/usr/include/OpenEXR -I/usr/include/openjpeg-2.4 -I/usr/include/orc-0.4 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -flto=auto -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu++20 -O2 -g -Werror=empty-body -Werror=pointer-arith -Werror=missing-declarations -Werror=return-type -Werror=misleading-indentation -Werror=format-security -Werror=suggest-override -Wno-missing-field-initializers -Wno-error=missing-field-initializers -Wno-unused-parameter -Wno-error=unused-parameter -ftree-vectorize -fvect-cost-model=very-cheap -march=native -fPIC -isystem/usr/include/opencv4 -pthread -isystem/usr/include/eigen3 -DQT_CORE_LIB -isystem/usr/include/x86_64-linux-gnu/qt5/QtCore -isystem/usr/include/x86_64-linux-gnu/qt5 -MD -MQ src/datactl/libsyntalos-datactl.so.1.2.0.p/vipsutils.cpp.o -MF src/datactl/libsyntalos-datactl.so.1.2.0.p/vipsutils.cpp.o.d -o src/datactl/libsyntalos-datactl.so.1.2.0.p/vipsutils.cpp.o -c ../src/datactl/vipsutils.cpp
psutils.cpp.o -c ../src/datactl/vipsutils.cpp
In file included from ../src/datactl/vipsutils.cpp:20:
../src/datactl/vipsutils.h: In function ‘vips::VImage newVipsImage(int, int, int)’:
../src/datactl/vipsutils.h:56:23: error: static assertion failed: Unknown VIPS format
   56 |         static_assert(false, "Unknown VIPS format");
      |                       ^~~~~
[97/1097] Compiling C++ object contrib...ibimplot.a.p/implot_implot_items.cpp.o
ninja: build stopped: subcommand failed.

```
The vips version in the standard repo was <8.14, so I compiled libvips from source successfully. The output of vips --version in my console is:
```
(base) llorente@llorente-desktop:~/syntalos/build$ vips --version
vips-8.15.2
```
I had built a couple of dependencies from source since their versions were too outdated, and I'm running this on Ubuntu 22.04. Any help would be welcome!

ximion, 2024-07-10:
This is an extremely strange error, because this particular template is not instantiated anywhere, so there is no way for the compile-time check to be hit - which C++ compiler are you using?

ximion, 2024-07-10:
@athp18 Can you check if the current change in master has resolved this issue for you?

athp18, 2024-07-12:
> @athp18 Can you check if the current change in master has resolved this issue for you?

It didn't unfortunately. I also reproduced the error on another Ubuntu PC. The only thing I can thing of is that it's maybe a compiler version -- I'm using GCC 12 although the default gcc-related packages in apt for 22.04 are for version 11.4, so I will try tomorrow to compile with Clang and let you know. Thank you for all your help!

ximion, 2024-07-12:
GCC should work and is preferred - this is super strange though, as the CI builds the code on Ubuntu 24.04 without any issues...

ximion, 2024-07-12:
Wait, you wrote 22.04! It may be that the compiler is actually too old to handle some of the C++20 constructs... Is it feasible for you to upgrade to 24.04 LTS? The Iceoryx version in 22.04 should also actually be too old (which is why the Flatpak is preferred for people using 22.04 starting with the next release).

Versioning is a bit annoying, as we soon also have to migrate to Qt 6 which will be a problem with Ubuntu 24.04 due to missing libraries. But the aim is to always support at least the last Ubuntu LTS release (and the Flatpak build of course, which should work almost anywhere).

athp18, 2024-07-12:
Ah, that would make sense! I will test it out on 24.04 and let you know if the issue goes away.

athp18, 2024-07-12:
Yep, updating to 24.04 solved the problem! Thanks for all your help!

closed 2024-07-12 by athp18