syntalos/syntalos #8: Unable to build

SpindleMS, 2024-07-29:
Good morning,

I've installed Syntalos via Flathub and it works perfectly on Ubuntu 24.04.  I'm now trying to build it to include the camera-flir module and I'm running into problems.

When I try to add the Syntalos PPA I get the following error:

_Ign:7 https://ppa.launchpadcontent.net/ximion/syntalos/ubuntu noble InRelease
Err:8 https://ppa.launchpadcontent.net/ximion/syntalos/ubuntu noble Release
  404  Not Found [IP: 185.125.190.80 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/ximion/syntalos/ubuntu noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default._

When I try to build using meson I get:

_Run-time dependency opencl found: NO (tried pkgconfig and cmake)

../meson.build:223:13: ERROR: Dependency "OpenCL" not found, tried pkgconfig and cmake_

I've verified that I have opencl installed so I'm not sure where this is coming from.

Thank you for your help

ximion, 2024-07-29:
Hi!
On Ubuntu 24.04, you will not need the PPA :-)

What you do need however are the OpenCL headers in the package `ocl-icd-opencl-dev`.
You can install all dependencies with this command:

```bash
sudo apt install -yq --no-install-recommends \
    gdb gcc g++ \
    cmake \
    gettext \
    libaravis-dev \
    libavcodec-dev \
    libavformat-dev \
    libavutil-dev \
    libegl-dev \
    libeigen3-dev \
    libglib2.0-dev \
    libgstreamer1.0-dev \
    libgstreamer-plugins-base1.0-dev \
    libkf5archive-dev \
    libkf5texteditor-dev \
    libkf5windowsystem-dev \
    libqtermwidget5-1-dev \
    libvips-dev \
    libopencv-dev \
    libpipewire-0.3-dev \
    libqt5opengl5-dev \
    libiceoryx-introspection-dev \
    libqt5serialport5-dev \
    libqt5svg5-dev \
    libswscale-dev \
    libusb-1.0-0-dev \
    libv4l-dev \
    libxml2-dev \
    libxxhash-dev \
    libsystemd-dev \
    systemd-dev \
    meson \
    ninja-build \
    ocl-icd-opencl-dev \
    pybind11-dev \
    python3-dev \
    python3-numpy \
    qtbase5-dev \
    qtmultimedia5-dev \
    udev \
    uuid-dev
```

Good luck and let me know if you have any other issues!

ximion, 2024-07-29:
Hi!
On Ubuntu 24.04, you will not need the PPA :-)

What you do need however are the OpenCL headers in the package `ocl-icd-opencl-dev`.
You can install all dependencies with this command:

```bash
sudo apt install --no-install-recommends \
    gdb gcc g++ \
    cmake \
    gettext \
    libaravis-dev \
    libavcodec-dev \
    libavformat-dev \
    libavutil-dev \
    libegl-dev \
    libeigen3-dev \
    libglib2.0-dev \
    libgstreamer1.0-dev \
    libgstreamer-plugins-base1.0-dev \
    libkf5archive-dev \
    libkf5texteditor-dev \
    libkf5windowsystem-dev \
    libqtermwidget5-1-dev \
    libvips-dev \
    libopencv-dev \
    libpipewire-0.3-dev \
    libqt5opengl5-dev \
    libiceoryx-introspection-dev \
    libqt5serialport5-dev \
    libqt5svg5-dev \
    libswscale-dev \
    libusb-1.0-0-dev \
    libv4l-dev \
    libxml2-dev \
    libxxhash-dev \
    libsystemd-dev \
    systemd-dev \
    meson \
    ninja-build \
    ocl-icd-opencl-dev \
    pybind11-dev \
    python3-dev \
    python3-numpy \
    qtbase5-dev \
    qtmultimedia5-dev \
    udev \
    uuid-dev
```

Good luck and let me know if you have any other issues!

closed 2024-07-30 by ximion

SpindleMS, 2024-07-30:
Thank you for getting back so quickly.  Your comment covered all of the required dependencies so that part worked out.  I did have to copy several header files into /src/fabric/streams.

I'm now running into compiler errors regarding flircameramod.cpp.  I'll attach a .txt file of the compiler output to keep the comment thread clean.

[camera-flir_compiler_error.txt](https://github.com/user-attachments/files/16431034/camera-flir_compiler_error.txt)

ximion, 2024-07-30:
Yeah, so, the FLIR Camera module is in huge disrepair... It would really need some serious work to reach the quality of the other camera modules, one major issue being that the Spinnaker SDK is closed-source and it is really hard to integrate into the existing test pipeline.

Plus, I currently don't have a FLIR camera at hand to properly test this.

So, since you are compiling the current Git master version anyway and FLIR cameras support the GenICam standard (for the most part), would you like to try the _Aravis Camera_ module with your device instead?
Ideally it will work just fine, and should even give you access to pretty much all camera features. The drawback of this module currently is that it's very new and in active development, so you may encounter some rough edges (we have been testing it here for weeks though without finding any major issues besides cosmetic problems in the UI).

SpindleMS, 2024-07-31:
I'm up for trying it.  I'll test it out over the next few days and let you know if it works out.  Thanks

ximion, 2024-07-31:
Very good, I'm quite curious :-) I recently whitelisted FLIR cameras for this, so they should also show up now (alongside The Imaging Source and Basler devices).

Of course, if you want to, you could also fix up the original Spinnaker-FLIR module, patches are welcome! :-)

SpindleMS, 2024-07-31:
Awesome, thanks for putting this all together.  So far I'm feeling a bit out of my depth but hopefully that won't last for long